usbarmory/usbarmory-debian-base_image #9: Issue importing gpg key: gpg: keyserver receive failed: Cannot assign requested address

jcalfee, 2018-06-27:
I'm having an issue importing the key:
```bash
Step 4/8 : RUN gpg --keyserver hkp://keys.gnupg.net --recv-keys 38DBBDC86092693E
 ---> Running in 857af6ee308a
gpg: directory '/root/.gnupg' created
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: keyserver receive failed: Cannot assign requested address
```

I don't see a clear solution out there.

```Dockerfile
FROM debian:stretch

RUN apt update
RUN apt install -y bc binfmt-support bzip2 fakeroot gcc gcc-arm-linux-gnueabihf git gnupg make parted qemu-user-static wget xz-utils zip debootstrap
RUN gpg --keyserver hkp://keys.gnupg.net --recv-keys 38DBBDC86092693E
RUN gpg --keyserver hkp://keys.gnupg.net --recv-keys 87F9F635D31D7652

RUN git clone https://github.com/inversepath/usbarmory-debian-base_image.git
WORKDIR /usbarmory-debian-base_image
RUN make all
```

But it works with gpg2:
```Dockerfile
RUN apt install -y gnupg2
RUN gpg2 --keyserver hkp://keys.gnupg.net --recv-keys 38DBBDC86092693E
```

Related: https://github.com/rvm/rvm/issues/3544

Feel free to close this issue.  I'm mostly documenting the error.

andrejro, 2018-07-02:
Hi!

I think this is only a Debian 8 related issue where gnupg binds to gnupg1. On Debian 9,
which is the one we support, the gnupg binds to gnupg2.
Instead, we found that dirmngr package should be added to the list of dependancies.

Can you just confirm that you are using Debian 8?

jcalfee, 2018-07-03:
It is debian:stretch (Debian 9)
```Dockerfile
FROM debian:stretch
...
```

andrejro, 2018-07-04:
Sorry, I missed that.
But in Debian 9 the gpg binary is gpg2:

/usr/bin/gpg --version
gpg (GnuPG) 2.1.18
libgcrypt 1.7.6-beta
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /root/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2

Not sure what happen in your case, maybe for some reason dirmngr was not installed and the installation of gnupg2 package installed it as a dependence. Now we have documented the dirmngr requirement explicitly.

Thanks.

closed 2018-07-04 by andrejro

jmendeth, 2018-12-22:
This just happened to me in an unrelated container (docker/php; same setup, stretch and gnupg installed), and [enabling IPv6 support in Docker](https://docs.docker.com/config/daemon/ipv6/) allows GPG to bind and succeed even if you don't have IPv6 internet.

(Commenting here mainly because this is the first Google result for the error)

lentschi, 2019-01-05:
For me adding the following to the Dockerfile did the trick:

```Dockerfile
RUN mkdir ~/.gnupg
RUN echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
```

The advantage to jmendeth's solution is that this doesn't require reconfiguring docker itself, saving users of the Dockerfile to take this extra preinstallation step. (The disadvantage is obvious: The Dockerfile needs to be changed.)

See https://rvm.io/rvm/security#ipv6-issues

(Like jmendeth I'm just commenting here because this was the first hit on Google. To me the error happened with a fresh `FROM ubuntu:18.04` docker build)

ericsubach, 2019-01-10:
lentschi's solution worked for me. I too was getting errors from fresh builds.

ElijahLynn, 2019-02-23:
> For me adding the following to the Dockerfile did the trick:
>
> ```dockerfile
> RUN mkdir ~/.gnupg
> RUN echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
> ```
>
> The advantage to jmendeth's solution is that this [...]

For those testing this with `docker run`, the rvm.io link mentions checking for existing processes. In my case I had already run the command once before so `dirmngr` was running which I `kill`d then re-ran command and it worked. I was stumped as I did have the ipv6 disabling catted out.

ElijahLynn, 2019-02-23:
> For me adding the following to the Dockerfile did the trick:
>
> ```dockerfile
> RUN mkdir ~/.gnupg
> RUN echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
> ```
>
> The advantage to jmendeth's solution is that this [...]

Since this is a top result in Google, and for newer Docker users this should really be combined into your existing RUN command if possible and at the very least squashed like this:

> RUN mkdir ~/.gnupg && echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf

As it will be one less cache layer then and faster.

dshiryaev-plesk, 2019-06-04:
I solved the problem by replacing `gpg --keyserver` with `gpg --batch --keyserver`.

blak3mill3r, 2019-06-05:
I solved this by restarting the Docker daemon (host machine is Ubuntu Xenial)

MichaelBrenden, 2019-06-16:
As mentioned above, this worked for me also:

RUN mkdir ~/.gnupg
RUN echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf

But to overcome another, intermittent problem (name resolution failure, due to slow DNS response), Dockerfile also needed this --

RUN ping -c 10 download.redis.io && ping -c 10 pool.sks-keyservers.net

webbnh, 2019-07-08:
> RUN mkdir ~/.gnupg && echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf

It's subtle, but I needed the following, instead:
```
RUN mkdir ~/.gnupg || echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
```
(In my case, the container already had the directory.)

orf, 2019-07-11:
`mkdir -p ~/.gnupg` is what you needed, not `||`

michele-amati, 2019-09-10:
> For me adding the following to the Dockerfile did the trick:
>
> ```dockerfile
> RUN mkdir ~/.gnupg
> RUN echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
> ```
>
> The advantage to jmendeth's solution is that this [...]

Had a similar problem, but with `apt-key adv` and since google brought me here I will leave my findings to ease the life of whoever comes next.

This solution works, but if you have this problem while calling `gpg` though `apt-key adv` you should know that the config file `~/.gnupg/dirmngr.conf` (where you disable ipv6) is ignored unless you explicitly specify it with the  `--homedir` param, example:

```
apt-key adv --homedir ~/.gnupg --keyserver <something> --recv-keys <something>
```

gruaig, 2019-11-14:
RUN mkdir ~/.gnupg && echo "disable-ipv6" >> ~/.gnupg/dirmngr.conf
Works thanks to @lentschi :D

luckydonald, 2020-04-26:
If you want to use a temporary folder for that (to not end up with that file in the finished container), you can do
```bash
export GNUPGHOME="$(mktemp -d)" &&  echo "disable-ipv6" >> ${GNUPGHOME}/dirmngr.conf`
```
(Add `RUN` in front of it, or add it to something existing.)

`mktemp -d` also replaces the `mkdir ~/.gnupg` step.