usbarmory/usbarmory-debian-base_image #19: Mk I: u-boot-2019.07 Removed

eduncan911, 2019-10-08:
Getting an error on `make all V=mark-one IMX=imx53` during the finalizer step:

```
sudo dd if=u-boot-2019.07/u-boot-dtb.imx of=usbarmory-mark-one-debian_stretch-base_image-20191008.raw bs=512 seek=2 conv=fsync conv=notrunc
dd: failed to open 'u-boot-2019.07/u-boot-dtb.imx': No such file or directory
Makefile:230: recipe for target 'finalize' failed
make: *** [finalize] Error 1
```

The `dtb` or Device Tree Blob, seems to be missing from the `u-boot` build process.  The normal `u-boot.imx` is there.  

Maybe it was renamed?  

```
root@827813faa247:/opt/armory/u-boot-2019.07# ls -la u-boot*
-rwxr-xr-x 1 root root 3186484 Oct  8 08:32 u-boot
-rwxr-xr-x 1 root root  382628 Oct  8 08:32 u-boot-nodtb.bin
-rwxr-xr-x 1 root root  382628 Oct  8 08:32 u-boot.bin
-rw-r--r-- 1 root root    8124 Oct  8 08:31 u-boot.cfg
-rw-r--r-- 1 root root    4669 Oct  8 08:32 u-boot.cfg.configs
-rw-r--r-- 1 root root    1842 Oct  8 08:32 u-boot.cfgout
-rw-r--r-- 1 root root  388096 Oct  8 08:32 u-boot.imx
-rw-r--r-- 1 root root     194 Oct  8 08:32 u-boot.imx.log
-rw-r--r-- 1 root root    1719 Oct  8 08:32 u-boot.lds
-rw-r--r-- 1 root root  473023 Oct  8 08:32 u-boot.map
-rwxr-xr-x 1 root root 1148026 Oct  8 08:32 u-boot.srec
-rw-r--r-- 1 root root  119664 Oct  8 08:32 u-boot.sym

abarisani, 2019-10-08:
This should be now fixed, please re-open if necessary. Thanks!

closed 2019-10-08 by abarisani

eduncan911, 2019-10-08:
Confirmed the `finalizer` works now on Mk I.  Thanks!