usbarmory/usbarmory-debian-base_image #22: Would you like an C.I. to test the builds?

eduncan911, 2019-10-08:
With the addition of the Dockerfile in #21 , I can put together a Travis or maybe even a Github Tasks build environment test for and report the state of `master`, and all PRs.

I have GitHub Tasks enabled in my account; however, I think everyone has to request enrollment.  

Let me know and I will try to whip it up.  Most likely will test all 3 build targets of the Makefile, and test that the files exists with no build errors.

abarisani, 2019-10-08:
I tried Travis already (see https://github.com/inversepath/usbarmory-debian-base_image/commit/dc3b758893db90f2d7f01c8c61bd297f98b23082), the problem is that the build takes too long and so it is not compatible with Travis time limits.

eduncan911, 2019-10-08:
Ah yes, it does take a long time.

Let me see what the restrictions are around GitHub Tasks as I've forked it over there on my accounts.

eduncan911, 2019-10-11:
I'm making some good headway with Github Actions (Github's native, free, unlimited CICD tool).

https://github.com/features/actions

I'll start on the usbarmory builds this weekend or next week.

eduncan911, 2019-10-12:
Almost done.  I'm getting an error running on Ubuntu 18.04 and compiling for the Mark II.  (Mark I passes)

Could you take a look?  Maybe no one has built this on Ubuntu 18.04?  Maybe there's something I can tweak?

https://github.com/eduncan911/usbarmory-debian-base_image/runs/257692569#step:9:115

Debian is not an option to build on this platform.  I also tried Debian9 in the docker image to build; however, I am unable to pass build commands dynamically to the docker image from within their C.I..  Decided to abandon it and just build directly on their system.  However, they only have Ubuntu 16.04 and Ubuntu 18.04 available for Linux builds.

abarisani, 2019-10-13:
The README highlights some keys that need to be imported in the pre-requisites section, have you imported that in your action?

eduncan911, 2019-10-13:
> The README highlights some keys that need to be imported in the pre-requisites section, have you imported that in your action?

Yes, I was attempting to - and it does import both in the other runs.  But now I see only 1 got imported during that failed `Import Signing Keys` step.  

```
retry=0; until [ $retry -ge 5 ]; do
    gpg --keyserver hkp://keys.gnupg.net --recv-keys 38DBBDC86092693E 87F9F635D31D7652 && break
    n=$[$n+1]
    sleep $((1 + RANDOM % 30))
done
shell: /bin/bash -e {0}
gpg: directory '/home/runner/.gnupg' created
gpg: keybox '/home/runner/.gnupg/pubring.kbx' created
gpg: key 38DBBDC86092693E: 2 duplicate signatures removed
gpg: key 38DBBDC86092693E: 174 signatures not checked due to missing keys
gpg: /home/runner/.gnupg/trustdb.gpg: trustdb created
gpg: key 38DBBDC86092693E: public key "Greg Kroah-Hartman <gregkh@linuxfoundation.org>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 1
gpg:               imported: 1
```

So odd, the other jobs actually did import both:

```
retry=0; until [ $retry -ge 5 ]; do
    gpg --keyserver hkp://keys.gnupg.net --recv-keys 38DBBDC86092693E 87F9F635D31D7652 && break
    n=$[$n+1]
    sleep $((1 + RANDOM % 30))
done
shell: /bin/bash -e {0}
gpg: directory '/home/runner/.gnupg' created
gpg: keybox '/home/runner/.gnupg/pubring.kbx' created
gpg: key 87F9F635D31D7652: 2 signatures not checked due to missing keys
gpg: /home/runner/.gnupg/trustdb.gpg: trustdb created
gpg: key 87F9F635D31D7652: public key "Thomas Rini <trini@konsulko.com>" imported
gpg: key 38DBBDC86092693E: 2 duplicate signatures removed
gpg: key 38DBBDC86092693E: 174 signatures not checked due to missing keys
gpg: key 38DBBDC86092693E: public key "Greg Kroah-Hartman <gregkh@linuxfoundation.org>" imported
gpg: no ultimately trusted keys found
gpg: Total number processed: 2
gpg:               imported: 2
```

Humm.  That doesn't address the `config` target missing in the other build/link, since it did import 2 PGP keys there.

eduncan911, 2019-10-13:
Thanks @abarisani.  Yep, I was importing but had a bug in my retry logic (as PGP keyservers are unreliable).  Fixed and now they always get both keys.

Switching to `make all` seems to have resolved it all as well!  Across all 5 builds (i added `imx6ul` and `imx6ull` as separate builds).  So we're ready to go!

I'll clean up the branch and will submit a PR.  However, next steps is that **you need to file a request to get access to Github Actions to enable it for your Github Org** or it will never work:

https://github.com/features/actions

---

More info...

I was previously building each `make` target, so it's easier to debug.  However, using the individual targets kept giving the errors during the `finalizer` target mentioned earlier.  `u-boot` config and target issues.  I suspect the Makefile needs some tweaking to allow for mk-ii individual target builds.

```
    - run: make V=${{ matrix.device }} IMX=${{ matrix.arch }} BOOT=${{ matrix.boot }} check_version
    - run: make V=${{ matrix.device }} IMX=${{ matrix.arch }} BOOT=${{ matrix.boot }} linux-deb
    - run: make V=${{ matrix.device }} IMX=${{ matrix.arch }} BOOT=${{ matrix.boot }} debian 
    - run: make V=${{ matrix.device }} IMX=${{ matrix.arch }} BOOT=${{ matrix.boot }} u-boot 
    - run: make V=${{ matrix.device }} IMX=${{ matrix.arch }} BOOT=${{ matrix.boot }} finalize
```

Besides, having so many steps would actually make things more fragile if we were ever to change the Makefile, which I'm assuming is still in flux? :)

abarisani, 2019-10-13:
Thanks for all the effort! I just signed up for Actions beta for our org, we will review your commits while we wait on that! Cheers.

abarisani, 2019-10-18:
Ok, I now have Actions enabled.

closed 2019-10-18 by abarisani

eduncan911, 2019-10-18:
Nice!

https://github.com/inversepath/usbarmory-debian-base_image/actions

abarisani, 2019-10-20:
I wonder if something like this can be used to directly upload to a new release?

https://github.com/JasonEtco/upload-to-release/blob/master/upload-to-release

It would be nice if the release would remain a draft, so that we can download and check the image and manually set it to non draft (the linked script does the opposite).

eduncan911, 2019-10-21:
There already an action for it.
https://github.com/eduncan911/usbarmory-debian-base_image/blob/b6d507b4f9f91e13e5989f270b3c683914e103f2/.github/workflows/tag-to-release.yml#L96-L104

Though, I currently have an issue open to see if they would do the work
first:  https://github.com/actions/upload-release-asset/issues/4

I thought you wanted to verify each binary first?  :)

The biggest hold up for me to submit a PR for it are your rich-text heavy
detailed Release Notes.  I've come up with a way to template-and-parse
release notes; just, I haven't had the time to work on it yet.

But yes, it can build all 3 versions.  Which, is kind of a problem...

I added a target to the Makefile for releases, which creates the sha256 of
each file:
https://github.com/eduncan911/usbarmory-debian-base_image/blob/b6d507b4f9f91e13e5989f270b3c683914e103f2/Makefile#L241-L243

However, this creates 12 files to upload!

I wanted to ask if you'd be willing to drop the XZ compression?  We'd then
build and upload only 6 files then.

abarisani, 2019-10-21:
I do want to verify each binary first, which is why I mentioned that I would be happy with upload to *draft* releases only. Because this allows me to download the published file on the draft release (which won't be public), validate it and then mark manually the release as non-draft. Does that make sense?

I'd like to peak only two releases, imx53 on uSD and imx6ULL on uSD, do you think this is feasible?

Thanks!

eduncan911, 2019-10-21:
Oh yes, easy!

The only hold up is no current action supports a body not multiple
attachments.

But if you want just two releases, I can specify just those files for now
to upload pretty quick.  :)

What about xz?  Keeping them?

mark-one-usd-...zip
mark-one-usd-...zip.sha256
mark-one-usd-...xz
mark-one-usd-...xz.sha256
mark-two-usd-...zip
mark-two-usd-...zip.sha256
mark-two-usd-...xz
mark-two-usd-...xz.sha256

That's 8 files.  If we drop xz, then it's only 4 files.

abarisani, 2019-10-21:
Ideally it would be one release with two attachments, but if that's not possible then two separate releases is fine. I'd keep both .zip and .xz files consistently with previous releases.

eduncan911, 2019-10-21:
Oh, it can be anything you like. One release with multiple attachments (8
attachments).  Two releases.

One release is best I think.  The large number files in the one release is
for the sha256 checksum I'm attaching, for each file.  I think it's best to
embed the file next to the release files.  Makes for automated parsing and
verification scripts.

Question: did you notice I've slightly renamed the filenames?  I had to add
"usd" and "emmc" to the naming convention so they could be separate in the
same build directory to upload from. It's all in that PR above I am testing
with.

I'm travelling today for work so I'll try to tinker with it on the plane
to/from.

abarisani, 2019-10-21:
I was thinking of one release with just 2 .xz files (uSD for mark-one and mark-two) and SHA256s, but if there are no constraints on Github why not making all valid combinations (uSD for mark-one and mark-two, eMMC for mark-two, SHA256 file and .xz/.zip for all) ?

eduncan911, 2019-10-21:
Heh, that's what I was originally proposing: attach all to one release, including sha hash files.  :)

I'll try to whip up something this week.

Note one restriction with GitHub Actions: their triggers for builds are kind of limited.  

So I was going to trigger a "Release" purely from a Tag.  You make a Tag, and push the tag up, and it will trigger a DRAFT release build and push all files up.

Note x2: since I have to build all of this in one pipeline, I have to do it synchronously.  That is, one at a time.  So the build will take about 1.5 to 2 hours to produce the final Release.

abarisani, 2019-10-21:
Ok, and sorry for the ping-pong. I was thinking out loud ;).