usbarmory/usbarmory-debian-base_image #25: Old golang version

jleni, 2020-01-15:
The go version that is installed in the docker container is very old (`go version go1.7.4 linux/amd64`) and there is not support for `go mod`.

jleni, 2020-01-15:
I can confirm. Manually installing go 1.13 inside the container fixed the problem.

abarisani, 2020-01-15:
There is now a package for armoryctl, you can just `apt-get update && apt-get install armoryctl`. Please confirm and close.

jleni, 2020-01-15:
The problem is that calling `make` inside the docker tries to compile it and then fails.. so it is not possible to build the debian image inside the container.
Installing a newer golang version in docker fixed it for me, so no worries, but I thought you might be interested to know.

abarisani, 2020-01-16:
Ah now I get it, will fix.