usbarmory/usbarmory-debian-base_image #28: can not boot from eMMC

peko19, 2020-02-19:
the same image is booting from microSD card with no problem, but from mmc got this on serial:

Card did not respond to voltage select!
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Card did not respond to voltage select!
Card did not respond to voltage select!
zimage: Bad magic!

abarisani, 2020-02-19:
The uSD and eMMC images require different configuration, how did you image the eMMC card?

Thanks

peko19, 2020-02-19:
i just used the one precompiled one and flashed it in the UMS mode with dd and later with etcher too.
please could i somewhere get a precompiled bootloader?

abarisani, 2020-02-20:
The pre-compiled image is only for uSD. For eMMC you have to build your own as shown at https://github.com/f-secure-foundry/usbarmory-debian-base_image

closed 2020-02-20 by abarisani