usbarmory/usbarmory-debian-base_image #31: Enable HTTPS on keys.inversepath.com used to fetch software signing keys

evilaliv3, 2020-04-14:
While starting experimenting with the packaging of [GlobaLeaks](https://github.com/globaleaks/GlobaLeaks/issues/2803) for USB Armory MKII i just found out that the GPG key used for verifying fetched resources is fetched via the plain HTTP resource: http://keys.inversepath.com/gpg-andrej.asc

I already produced a patch to be integrated in the repository but it seems to me that the server hosting the keys is not running HTTPS so that before integrating the patch it would be probably required to enable it.

In the meantime probably as an alternative could be hosted inside the script directly or served via the repository/github itself.

abarisani, 2020-04-14:
Thanks, we will address this ASAP.

danbia, 2020-04-14:
TLS has been enabled on our key server, you can download the key on:

https://keys.inversepath.com/gpg-andrej.asc

Thanks

closed 2020-04-14 by danbia