usbarmory/usbarmory-debian-base_image #35: Released and document build for the differnet versions of MKII and for MKI

evilaliv3, 2020-04-22:
I suggest that we could probably work on retesting it to release and document a version of Debian for all both the MKII platforms and as well for the MKI. This would help users to experiment and before producing their own image. I will try to provide my feedback and support for the retesting on your official repository.

I discovered in fact while testing a board with the packaging of GlobaLeaks that at least two different version of the MKII exists (https://github.com/globaleaks/GlobaLeaks/issues/2803) and that the two requires different build flags and kernel options.

abarisani, 2020-04-22:
The second variant of the Mk II is only for custom/bulk order and rarely seen in the wild, it doesn't end up in consumer hands. Therefore you can ignore it for your purposes.

evilaliv3, 2020-04-22:
I see, thank you @abarisani for the explaination

andrejro, 2020-07-15:
The build of all the supported USB armory images has been added.
https://github.com/f-secure-foundry/usbarmory-debian-base_image/releases/tag/20200714
Thanks @eduncan911 for your help!

closed 2020-07-15 by andrejro

eduncan911, 2020-07-15:
@andrejro I see you forked my version of the release action:

https://github.com/f-secure-foundry/usbarmory-debian-base_image/blob/master/.github/workflows/release.yml#L85

Looks like you missed one:

https://github.com/f-secure-foundry/usbarmory-debian-base_image/blob/master/.github/workflows/release.yml#L47

And I agree, Fsecure should own the entire toolchain. You should most likely fork the other tools as well, like this one:

https://github.com/f-secure-foundry/usbarmory-debian-base_image/blob/master/.github/workflows/release.yml#L23

As well as make, sign, and publish a custom Docker image to perform the builds (Actions can run Docker within Docker).

---

FYI, there aren't any changes in my Release version from the one I forked from.  I was trying to modify it to take larger files; but, i couldn't find an API that would let me.

andrejro, 2020-07-15:
Thanks for spotting this. Should be fixed now.
We will think if it is the case to use a custom Docker image. Thanks!

eduncan911, 2020-07-15:
Yeah, right now you are relying on a docker image as the build environment (https://github.com/f-secure-foundry/usbarmory-debian-base_image/blob/master/.github/workflows/release.yml#L15), which I haven't been able to find the source of yet.

And also, you are relying on an NodeJS script in a public GitHub repo to pull down the code from GitHub within (`actions/checkout@v2`).  

Just stating that there are some build-chain concerns that could be closed on with a custom Docker image signed by Fsecure, performing the actions instead.