usbarmory/usbarmory-debian-base_image #36: Make command results in error "losetup: /dev/loop0: failed to set up loop device: No such file or directory"

evilaliv3, 2020-04-22:
Following the guide for building the image while running the command ```make all V=mark-two IMX=imx6ulz BOOT=uSD``` i get the following error:

```losetup: /dev/loop0: failed to set up loop device: No such file or directory```

I receive the error at the first time i run the command while any further attempt result in a success.

I successfully reproduced the issue many times with a new ```git clone``` but i've still not understood the reason of the failure.

abarisani, 2020-04-22:
Most of the times this is caused by the `loop` module not being loaded.

abarisani, 2020-04-22:
Clarified in https://github.com/f-secure-foundry/usbarmory-debian-base_image/commit/58a3806cc67e949018c37710798334b50fe5c1d1

closed 2020-04-22 by abarisani