usbarmory/usbarmory-debian-base_image #39: Cant build in docker

vonProteus, 2020-09-22:
i'm trying to build it on my mac (10.15.6) with docker
i'm checkout to 20200714

```
$ docker info
Client:
 Debug Mode: false

Server:
 Containers: 9
  Running: 0
  Paused: 0
  Stopped: 9
 Images: 65
 Server Version: 19.03.12
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc version: dc9208a3303feef5b3839f4323d9beb36df0a9dd
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.19.76-linuxkit
 Operating System: Docker Desktop
 OSType: linux
 Architecture: x86_64
 CPUs: 2
 Total Memory: 1.945GiB
 Name: docker-desktop
 ID: ELHR:33LN:BXHM:R3OB:PQRH:VVKC:AC74:2L55:RW3K:QMTM:X7FD:2OYJ
 Docker Root Dir: /var/lib/docker
 Debug Mode: true
  File Descriptors: 40
  Goroutines: 52
  System Time: 2020-09-22T12:09:32.558559522Z
  EventsListeners: 3
 HTTP Proxy: gateway.docker.internal:3128
 HTTPS Proxy: gateway.docker.internal:3129
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
 Product License: Community Engine
```

```
$ docker build --rm -t armory ./  
Sending build context to Docker daemon  195.1kB
Step 1/11 : FROM debian:10
 ---> f6dcff9b59af
Step 2/11 : RUN apt-get update && apt-get upgrade -y
 ---> Using cache
 ---> 9cc71a570877
Step 3/11 : RUN apt-get install -y     bc binfmt-support bzip2 fakeroot gcc gcc-arm-linux-gnueabihf     git gnupg make parted qemu-user-static wget xz-utils zip     debootstrap sudo dirmngr bison flex libssl-dev kmod udev cpio
 ---> Using cache
[... 229 lines cut ...]
Length: 989 [text/plain]
Saving to: 'linux-5.4.51.tar.sign'

linux-5.4.51.tar.sign                                              100%[===============================================================================================================================================================>]     989  --.-KB/s    in 0.002s

2020-09-22 12:18:15 (408 KB/s) - 'linux-5.4.51.tar.sign' saved [989/989]

gpg: assuming signed data in 'linux-5.4.51.tar'
gpg: Signature made Thu Jul  9 07:39:22 2020 UTC
gpg:                using RSA key 647F28654894E3BD457199BE38DBBDC86092693E
gpg: Good signature from "Greg Kroah-Hartman <gregkh@linuxfoundation.org>" [unknown]
gpg:                 aka "Greg Kroah-Hartman <gregkh@kernel.org>" [unknown]
gpg:                 aka "Greg Kroah-Hartman (Linux kernel stable release signing key) <greg@kroah.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 647F 2865 4894 E3BD 4571  99BE 38DB BDC8 6092 693E
tar: linux-5.4.51/Documentation/Changes: Cannot utime: No such file or directory
tar: linux-5.4.51/arch/arm/boot/dts/sun8i-a23-ippo-q8h-v1.2.dts: Cannot utime: No such file or directory
tar: linux-5.4.51/arch/arm/boot/dts/sun8i-a23-ippo-q8h-v5.dts: Cannot utime: No such file or directory
tar: linux-5.4.51/arch/arm/boot/dts/sun8i-a33-et-q8-v1.6.dts: Cannot utime: No such file or directory
tar: linux-5.4.51/arch/arm/boot/dts/sun8i-a33-ippo-q8h-v1.2.dts: Cannot utime: No such file or directory
tar: Exiting with failure status due to previous errors
make: *** [Makefile:124: linux-5.4.51/arch/arm/boot/zImage] Error 2
root@88fa4924a1dc:/opt/armory#
```

abarisani, 2020-09-22:
This seems to be a Docker bug under macOS, can you try modifying the Makefile so that the `tar` command has also the `-m` flag ?

vonProteus, 2020-09-22:
my modification
```
diff --git i/Makefile w/Makefile
index c3857df..dc8525b 100644
--- i/Makefile
+++ w/Makefile
@@ -124,7 +124,7 @@ linux-${LINUX_VER}/arch/arm/boot/zImage: check_version linux-${LINUX_VER}.tar.xz
        @if [ ! -d "linux-${LINUX_VER}" ]; then \
                unxz --keep linux-${LINUX_VER}.tar.xz; \
                gpg --verify linux-${LINUX_VER}.tar.sign; \
-               tar xf linux-${LINUX_VER}.tar && cd linux-${LINUX_VER}; \
+               tar -m xf linux-${LINUX_VER}.tar && cd linux-${LINUX_VER}; \
        fi
        wget ${USBARMORY_REPO}/software/kernel_conf/${V}/usbarmory_linux-${LINUX_VER_MAJOR}.config -O linux-${LINUX_VER}/.config
        if test "${V}" = "mark-two"; then \
@@ -139,7 +139,7 @@ linux-${LINUX_VER}/arch/arm/boot/zImage: check_version linux-${LINUX_VER}.tar.xz

 u-boot-${UBOOT_VER}/u-boot.bin: check_version u-boot-${UBOOT_VER}.tar.bz2
        gpg --verify u-boot-${UBOOT_VER}.tar.bz2.sig
-       tar xf u-boot-${UBOOT_VER}.tar.bz2
+       tar -m xf u-boot-${UBOOT_VER}.tar.bz2
        cd u-boot-${UBOOT_VER} && make distclean
        @if test "${V}" = "mark-one"; then \
                cd u-boot-${UBOOT_VER} && make usbarmory_config; \
```
and result 

```                   
root@ff251b508b33:/opt/armory# make all V=mark-two IMX=imx6ulz BOOT=eMMC
target: USB armory V=mark-two IMX=imx6ulz BOOT=eMMC
--2020-09-22 13:16:55--  https://github.com/f-secure-foundry/armoryctl/archive/v1.0.zip
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/f-secure-foundry/armoryctl/zip/v1.0 [following]
--2020-09-22 13:16:55--  https://codeload.github.com/f-secure-foundry/armoryctl/zip/v1.0
Resolving codeload.github.com (codeload.github.com)... 140.82.121.10
Connecting to codeload.github.com (codeload.github.com)|140.82.121.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: 'armoryctl-v1.0.zip'

armoryctl-v1.0.zip                                                     [  <=>                                                                                                                                                           ]  24.62K  32.3KB/s    in 0.8s

2020-09-22 13:16:57 (32.3 KB/s) - 'armoryctl-v1.0.zip' saved [25206]

Archive:  armoryctl-v1.0.zip
7a563b78b9b78c201d326afd27afdd1f4a30a746
   creating: armoryctl-1.0/
  inflating: armoryctl-1.0/LICENSE
  inflating: armoryctl-1.0/Makefile
  inflating: armoryctl-1.0/README.md
   creating: armoryctl-1.0/anna_b112/
  inflating: armoryctl-1.0/anna_b112/at.go
  inflating: armoryctl-1.0/anna_b112/openocd.go
  inflating: armoryctl-1.0/armoryctl.go
   creating: armoryctl-1.0/atecc608a/
  inflating: armoryctl-1.0/atecc608a/atecc608a.go
   creating: armoryctl-1.0/fusb303/
  inflating: armoryctl-1.0/fusb303/fusb303.go
  inflating: armoryctl-1.0/go.mod
  inflating: armoryctl-1.0/go.sum
   creating: armoryctl-1.0/internal/
  inflating: armoryctl-1.0/internal/armoryctl.go
  inflating: armoryctl-1.0/internal/gpio.go
  inflating: armoryctl-1.0/internal/i2c.go
  inflating: armoryctl-1.0/internal/led.go
  inflating: armoryctl-1.0/internal/uart.go
  inflating: armoryctl-1.0/internal/util.go
   creating: armoryctl-1.0/led/
[... 151 lines cut ...]
--2020-09-22 13:17:53--  https://mirrors.edge.kernel.org/pub/linux/kernel/v5.x/linux-5.4.51.tar.sign
Resolving mirrors.edge.kernel.org (mirrors.edge.kernel.org)... 147.75.101.1, 2604:1380:2001:3900::1
Connecting to mirrors.edge.kernel.org (mirrors.edge.kernel.org)|147.75.101.1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 989 [text/plain]
Saving to: 'linux-5.4.51.tar.sign'

linux-5.4.51.tar.sign                                              100%[===============================================================================================================================================================>]     989  --.-KB/s    in 0.003s

2020-09-22 13:17:53 (315 KB/s) - 'linux-5.4.51.tar.sign' saved [989/989]

gpg: assuming signed data in 'linux-5.4.51.tar'
gpg: Signature made Thu Jul  9 07:39:22 2020 UTC
gpg:                using RSA key 647F28654894E3BD457199BE38DBBDC86092693E
gpg: Good signature from "Greg Kroah-Hartman <gregkh@linuxfoundation.org>" [unknown]
gpg:                 aka "Greg Kroah-Hartman <gregkh@kernel.org>" [unknown]
gpg:                 aka "Greg Kroah-Hartman (Linux kernel stable release signing key) <greg@kroah.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 647F 2865 4894 E3BD 4571  99BE 38DB BDC8 6092 693E
tar: You must specify one of the '-Acdtrux', '--delete' or '--test-label' options
Try 'tar --help' or 'tar --usage' for more information.
make: *** [Makefile:124: linux-5.4.51/arch/arm/boot/zImage] Error 2
root@ff251b508b33:/opt/armory#
```

abarisani, 2020-09-22:
I think you have to pass it as `tar mxf` in this case, therefore without mixing flag and flag-less options.

eduncan911, 2020-09-22:
For the record, when I tweaked the Dockerfile and Makefile in the past to build from my macOS I was using the GNU version of Tar, as I replace all macOS utils with GNU for consistency.

https://github.com/eduncan911/dotfiles/blob/master/.Brewfile#L45

However, I don't think that is the case here...  Because this is within the Docker environment - not macOS utils.

vonProteus, 2020-09-22:
my modification
```
diff --git i/Makefile w/Makefile
index c3857df..710cd2d 100644
--- i/Makefile
+++ w/Makefile
@@ -124,7 +124,7 @@ linux-${LINUX_VER}/arch/arm/boot/zImage: check_version linux-${LINUX_VER}.tar.xz
        @if [ ! -d "linux-${LINUX_VER}" ]; then \
                unxz --keep linux-${LINUX_VER}.tar.xz; \
                gpg --verify linux-${LINUX_VER}.tar.sign; \
-               tar xf linux-${LINUX_VER}.tar && cd linux-${LINUX_VER}; \
+               tar mxf linux-${LINUX_VER}.tar && cd linux-${LINUX_VER}; \
        fi
        wget ${USBARMORY_REPO}/software/kernel_conf/${V}/usbarmory_linux-${LINUX_VER_MAJOR}.config -O linux-${LINUX_VER}/.config
        if test "${V}" = "mark-two"; then \
@@ -139,7 +139,7 @@ linux-${LINUX_VER}/arch/arm/boot/zImage: check_version linux-${LINUX_VER}.tar.xz

 u-boot-${UBOOT_VER}/u-boot.bin: check_version u-boot-${UBOOT_VER}.tar.bz2
        gpg --verify u-boot-${UBOOT_VER}.tar.bz2.sig
-       tar xf u-boot-${UBOOT_VER}.tar.bz2
+       tar mxf u-boot-${UBOOT_VER}.tar.bz2
        cd u-boot-${UBOOT_VER} && make distclean
        @if test "${V}" = "mark-one"; then \
                cd u-boot-${UBOOT_VER} && make usbarmory_config; \
```
and result 
```
$ docker run -it --privileged -v $(pwd):/opt/armory --name armory --rm armory make all V=mark-two IMX=imx6ulz BOOT=eMMC

target: USB armory V=mark-two IMX=imx6ulz BOOT=eMMC
--2020-09-22 13:44:02--  https://github.com/f-secure-foundry/armoryctl/archive/v1.0.zip
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/f-secure-foundry/armoryctl/zip/v1.0 [following]
--2020-09-22 13:44:03--  https://codeload.github.com/f-secure-foundry/armoryctl/zip/v1.0
Resolving codeload.github.com (codeload.github.com)... 140.82.121.10
Connecting to codeload.github.com (codeload.github.com)|140.82.121.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: 'armoryctl-v1.0.zip'

armoryctl-v1.0.zip                                                     [ <=>                                                                                                                                                            ]  24.62K  --.-KB/s    in 0.08s

2020-09-22 13:44:03 (303 KB/s) - 'armoryctl-v1.0.zip' saved [25206]

Archive:  armoryctl-v1.0.zip
7a563b78b9b78c201d326afd27afdd1f4a30a746
   creating: armoryctl-1.0/
  inflating: armoryctl-1.0/LICENSE
  inflating: armoryctl-1.0/Makefile
  inflating: armoryctl-1.0/README.md
   creating: armoryctl-1.0/anna_b112/
  inflating: armoryctl-1.0/anna_b112/at.go
  inflating: armoryctl-1.0/anna_b112/openocd.go
  inflating: armoryctl-1.0/armoryctl.go
   creating: armoryctl-1.0/atecc608a/
  inflating: armoryctl-1.0/atecc608a/atecc608a.go
   creating: armoryctl-1.0/fusb303/
  inflating: armoryctl-1.0/fusb303/fusb303.go
  inflating: armoryctl-1.0/go.mod
  inflating: armoryctl-1.0/go.sum
   creating: armoryctl-1.0/internal/
  inflating: armoryctl-1.0/internal/armoryctl.go
  inflating: armoryctl-1.0/internal/gpio.go
  inflating: armoryctl-1.0/internal/i2c.go
  inflating: armoryctl-1.0/internal/led.go
  inflating: armoryctl-1.0/internal/uart.go
  inflating: armoryctl-1.0/internal/util.go
   creating: armoryctl-1.0/led/
[... 2178 lines cut ...]
  LD [M]  drivers/usb/gadget/legacy/g_multi.o
  LD [M]  drivers/usb/gadget/legacy/g_ncm.o
  LD [M]  drivers/usb/gadget/legacy/g_acm_ms.o
  CC [M]  drivers/usb/gadget/epautoconf.o
  CC [M]  drivers/usb/gadget/composite.o
  CC [M]  drivers/usb/gadget/functions.o
  CC [M]  drivers/usb/gadget/configfs.o
  CC [M]  drivers/usb/roles/class.o
  LD [M]  drivers/usb/roles/roles.o
  CC [M]  drivers/usb/storage/scsiglue.o
  CC [M]  drivers/usb/gadget/u_f.o
  CC [M]  drivers/usb/storage/protocol.o
  LD [M]  drivers/usb/gadget/libcomposite.o
  CC [M]  drivers/usb/storage/transport.o
  CC [M]  drivers/usb/storage/usb.o
  CC [M]  drivers/usb/storage/initializers.o
  CC [M]  drivers/usb/storage/sierra_ms.o
  CC [M]  drivers/usb/storage/option_ms.o
  CC [M]  drivers/usb/storage/usual-tables.o
  LD [M]  drivers/usb/storage/usb-storage.o
  AR      drivers/usb/built-in.a
  AR      drivers/built-in.a
make[1]: Leaving directory '/opt/armory/linux-5.4.51'
make: *** [Makefile:127: linux-5.4.51/arch/arm/boot/zImage] Error 2
```

andrejro, 2020-09-29:
The issue is caused by the Mac's case-insensitive filesystem.
Looking for a workaround.

vonProteus, 2020-09-29:
yes i was about to write about it
my workaround is
create sparsebundle image in disk utility and format it with case sensitive file system 
i dont know how to fix it on windows

andrejro, 2020-09-30:
Added instruction on how to build on Mac OS X in 2489c04213d56616a146750ccd23e8e6c83eddde.

closed 2020-09-30 by andrejro