usbarmory/usbarmory-debian-base_image #42: Routing Internet access into Debian

bernardoaraujor, 2020-11-28:
I'm trying to provide internet access to my USB Armory (Debian) from my Ubuntu host.
I'm able to get an SSH shell just fine.

I'm running the [commands from the wiki](https://github.com/f-secure-foundry/usbarmory/wiki/Host-communication#setup--connection-sharing-linux), however I can't `ping 8.8.8.8` from the Armory.

cc @nothingismagick

abarisani, 2020-11-29:
Is your Internet facing interface named `wlan0` as assumed in the example? Do you have any firewall rules in place that my prevent USB armory traffic?

On Ubuntu the NetworkManager is quite invasive, it would be probably best to use Ubuntu GUI for this.

Something similar to [this](https://askubuntu.com/questions/169473/sharing-connection-to-other-pcs-via-wired-ethernet).

andrejro, 2020-12-11:
@bernardoaraujor is this issue still present?

bernardoaraujor, 2020-12-11:
sorry I haven't tried yet.
I'll give some feedback soon

bernardoaraujor, 2020-12-16:
issue solved.
solution was documented by @elenaf9 at https://github.com/iotaledger/stronghold.rs/issues/55#issuecomment-744405822

closed 2020-12-16 by bernardoaraujor