usbarmory/usbarmory-debian-base_image #43: xt_conntrack.ko kernel module missing from usbarmory-debian-base_image

fastjack77, 2021-01-01:
Good day,

Would it be possible to have the xt_conntrack.ko module added to the usbarmory-debian-base_image? I find ip_tables difficult to manage without certain modules. Or if there is an easy way to add these? Thank you!

```sudo modprobe xt_conntrack   
modprobe: FATAL: Module xt_conntrack not found in directory /lib/modules/5.4.82-0

usbarmory@usbarmory:~$ find /lib/modules -type f -name "*conntrack*"
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_snmp.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_h323.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_irc.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_netbios_ns.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_netlink.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_sane.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_tftp.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_sip.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_pptp.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_broadcast.ko
/lib/modules/5.4.82-0/kernel/net/netfilter/nf_conntrack_ftp.ko```

abarisani, 2021-01-07:
We will do so in the next release. Thanks!

andrejro, 2021-01-08:
Module added in https://github.com/f-secure-foundry/usbarmory-debian-base_image/releases/tag/20210108

closed 2021-01-08 by andrejro