usbarmory/usbarmory-debian-base_image #44: Unable to install Wireguard using DKMS

jeffhanson, 2021-01-15:
I am trying to get Wireguard installed on my USB Armory MkII running the latest Debian image. 

The wireguard packages are part of the buster-backports repository:
```
echo "deb http://deb.debian.org/debian buster-backports main" | sudo tee -a /etc/apt/sources.list
sudo apt update && sudo apt dist-upgrade && sudo apt install wireguard
...
Building for 5.4.87-0
Building initial module for 5.4.87-0
Error! Bad return status for module build on kernel: 5.4.87-0 (armv7l)
Consult /var/lib/dkms/wireguard/1.0.20201221/build/make.log for more information.
```
Here is the contents of the above make.log:
```
DKMS make.log for wireguard-1.0.20201221 for kernel 5.4.87-0 (armv7l)
Wed 13 Jan 2021 07:12:26 PM UTC
make: Entering directory '/usr/lib/modules/5.4.87-0/build'
make: *** No targets specified and no makefile found.  Stop.
make: Leaving directory '/usr/lib/modules/5.4.87-0/build'
```
I did some investigating and it appears the debian linux-headers package is normally built using the Linux kernel deb-pkg target which ends up calling `scripts/package/builddeb`. This script in turn makes a list of the files to be put in the package and then uses tar to put them in the right directory.

I created a script based on `builddeb` called `copy_header_files.sh` and updated the linux-headers-deb target as follows:
```
#### linux-headers-deb ####
HEADER_DEPS := check_version
HEADER_DEPS += linux-${LINUX_VER}/arch/arm/boot/zImage
linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf.deb: $(HEADER_DEPS)
        mkdir -p linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/{DEBIAN,boot,lib/modules/${LINUX_VER}${LOCALVERSION}}
        cd linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/lib/modules/${LINUX_VER}${LOCALVERSION} ; ln -sf /usr/src/linux-${LINUX_VER}${LOCALVERSION} build
        cat control_template_linux-headers | \
                sed -e 's/XXXX/${LINUX_VER_MAJOR}/'          | \
                sed -e 's/YYYY/${LINUX_VER}${LOCALVERSION}/' | \
                sed -e 's/ZZZZ/linux-image-${LINUX_VER_MAJOR}-usbarmory (=${LINUX_VER}${LOCALVERSION})/' | \
                sed -e 's/USB armory/USB armory ${V}/' \
                > linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/DEBIAN/control
        @if test "${V}" = "mark-two"; then \
                sed -i -e 's/${LINUX_VER_MAJOR}-usbarmory/${LINUX_VER_MAJOR}-usbarmory-mark-two/' \
                linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/DEBIAN/control; \
        fi
        # Build headers package similar to kernel script: scripts/package/builddeb
        ./copy_header_files.sh "linux-${LINUX_VER}" "linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/usr/src/linux-headers-${LINUX_VER}${LOCALVERSION}"
        chmod 755 linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf/DEBIAN
        fakeroot dpkg-deb -b linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf linux-headers-${LINUX_VER_MAJOR}-usbarmory-${V}_${LINUX_VER}${LOCALVERSION}_armhf.deb
```
Note the changed build link. The contents of the copy_header_files.sh:
```
#!/bin/sh
srctree=$1
destdir=$2

(cd $srctree; find . -name Makefile\* -o -name Kconfig\* -o -name \*.pl) > ../hdrsrcfiles
(cd $srctree; find arch/*/include include scripts -type f -o -type l) >> ../hdrsrcfiles
(cd $srctree; find arch/arm -name module.lds -o -name Kbuild.platforms -o -name Platform) >> ../hdrsrcfiles
(cd $srctree; find $(find arch/arm -name include -o -name scripts -type d) -type f) >> ../hdrsrcfiles
(cd $srctree; find arch/arm/include Module.symvers include scripts -type f) > ../hdrobjfiles
mkdir -p "$destdir"
(cd $srctree; tar -c -f - -T -) < "../hdrsrcfiles" | (cd $destdir; tar -xf -)
(cd $srctree; tar -c -f - -T -) < "../hdrobjfiles" | (cd $destdir; tar -xf -)
(cd $srctree; cp Kconfig ../$destdir/.config) # copy .config manually to be where it's expected to be
```

Running the `linux-headers-deb` target above results in a `linux-headers-5.4-usbarmory-mark-two_5.4.87-0_armhf.deb` that contains all the files needed to build the wireguard-dkms.

*HOWEVER*, a number of the binaries under the scripts directory are not cross-compiled. This appears to be a known issue when building for arm (see https://forum.loverpi.com/discussion/555/how-to-fix-dkms-error-bin-sh-1-scripts-basic-fixdep-exec-format-error and https://forum.radxa.com/t/dkms-building-error/307/5). This causes the dkms build to fail:
```
DKMS make.log for wireguard-1.0.20201221 for kernel 5.4.87-0-usbarmory (armv7l)
Fri 15 Jan 2021 02:45:34 AM UTC
make: Entering directory '/usr/src/linux-headers-5.4.87-0'
  AR      /var/lib/dkms/wireguard/1.0.20201221/build/built-in.a
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/main.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:262: /var/lib/dkms/wireguard/1.0.20201221/build/main.o] Error 2
make[1]: *** Deleting file '/var/lib/dkms/wireguard/1.0.20201221/build/main.o'
make: *** [Makefile:1732: /var/lib/dkms/wireguard/1.0.20201221/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.87-0'
```

I tried the recommended solution by re-building the scripts `cd /usr/src/linux-headers-5.4.87-0 && sudo make scripts` ON the usb armory, but this doesn't build all the tools needed and the DKMS build fails again:
```
DKMS make.log for wireguard-1.0.20201221 for kernel 5.4.87-0-usbarmory (armv7l)
Fri 15 Jan 2021 02:48:46 AM UTC
make: Entering directory '/usr/src/linux-headers-5.4.87-0'
  AR      /var/lib/dkms/wireguard/1.0.20201221/build/built-in.a
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/main.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/noise.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/device.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/peer.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/timers.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/queueing.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/send.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/receive.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/socket.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/peerlookup.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/allowedips.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/ratelimiter.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/cookie.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/netlink.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/chacha20/chacha20.o
  PERLASM /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/chacha20/chacha20-arm.S
  AS [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/chacha20/chacha20-arm.o
  AS [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/chacha20/chacha20-unrolled-arm.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/poly1305/poly1305.o
  PERLASM /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/poly1305/poly1305-arm.S
  AS [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/poly1305/poly1305-arm.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/chacha20poly1305.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/blake2s/blake2s.o
  CC [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/curve25519/curve25519.o
  AS [M]  /var/lib/dkms/wireguard/1.0.20201221/build/crypto/zinc/curve25519/curve25519-arm.o
  LD [M]  /var/lib/dkms/wireguard/1.0.20201221/build/wireguard.o
  Building modules, stage 2.
  MODPOST 1 modules
/bin/sh: 1: scripts/mod/modpost: Exec format error
make[1]: *** [scripts/Makefile.modpost:94: __modpost] Error 2
make: *** [Makefile:1645: modules] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.87-0'
```

I haven't yet figured out how to rebuild the modpost binary, and even if I do it doesn't solve the problem with there being x86_64 binaries in the linux-headers .deb file. At this point I may just try to build the deb files on an ARM system.

jeffhanson, 2021-01-16:
I was able to get this to work and have Wireguard running successfully. Unfortunately my solution isn't a good one.

After installing `qemu-user-static` on my host system I modified the Dockerfile to use an arm32v7 debian image:
```
FROM arm32v7/debian:10

RUN apt-get update && apt-get upgrade -y
RUN apt-get install -y \
    bc binfmt-support bzip2 fakeroot gcc gcc-arm-linux-gnueabihf \
    git gnupg make parted rsync qemu-user-static wget xz-utils zip \
    debootstrap sudo dirmngr bison flex libssl-dev kmod udev cpio

# import U-Boot signing keys
RUN gpg --batch --keyserver hkp://ha.pool.sks-keyservers.net --recv-keys 38DBBDC86092693E && \
    gpg --batch --keyserver hkp://ha.pool.sks-keyservers.net --recv-keys 147C39FF9634B72C && \
# import golang signing keys
    gpg --batch --keyserver hkp://ha.pool.sks-keyservers.net --recv-keys 7721F63BD38B4796

# install golang
ENV GOLANG_VERSION="1.15.6"
RUN wget -O go.tgz https://storage.googleapis.com/golang/go${GOLANG_VERSION}.linux-armv6l.tar.gz --progress=dot:giga
RUN wget -O go.tgz.asc https://storage.googleapis.com/golang/go${GOLANG_VERSION}.linux-armv6l.tar.gz.asc --progress=dot:giga
#RUN echo "3918e6cc85e7eaaa6f859f1bdbaac772e7a825b0eb423c63d3ae68b21f84b844 *go.tgz" | sha256sum --strict --check -
RUN gpg --batch --verify go.tgz.asc go.tgz
RUN tar -C /usr/local -xzf go.tgz && rm go.tgz

ENV PATH "$PATH:/usr/local/go/bin"
ENV GOPATH /go

WORKDIR /opt/armory
```

Build command: `docker build --no-cache --platform linux/arm/v7 -t armory-arm .`
Start container: `docker run --platform linux/arm/v7 --rm -it --privileged -v $(pwd):/opt/armory --name armory armor`

Once in the arm container I used the modified Makefile and copy_header_files.sh script mentioned above. I then built the headers package with: `make linux-headers-deb V=mark-two IMX=imx6ulz BOOT=eMMC`

Emulation of ARM on x86 is very slow so this took about 3 hours, but all of the binaries in the headers package were the right architecture. I installed the deb on the USB Armory after which I was able to complete the installation of the wireguard-dkms package and load wireguard module.

andrejro, 2022-06-28:
The `wireguard` package is now included in the Debian 11, it should be possible to use that one instead of `wireguard-dkms`.

closed 2022-06-28 by andrejro