usbarmory/usbarmory-debian-base_image #49: Where can the new public key be retrieved for the package repo?

assarbad, 2021-08-05:
Hi, when I attempt to look up `keys.gnupg.net` it fails and [your instructions](https://github.com/f-secure-foundry/usbarmory-debian-base_image#pre-requisites) also still reference the old key. But it appears a key change happened (or, perhaps less likely) your repository got compromised.

Where can one find the transition notice and the new public key?

Thanks,

Oliver

assarbad, 2021-08-05:
Meanwhile found https://github.com/f-secure-foundry/f-secure-foundry.github.io/commit/4f6dbd401dee82a1ae4342241570c3b3f3e5b3bd ... but I still can't make sense of the discrepancy.

eduncan911, 2021-08-05:
I believe the `Makefile` pulls the latest key on each build (it's part of the CICD here).  Does this help?

https://github.com/f-secure-foundry/usbarmory-debian-base_image/blob/d62df123b39af7b64c44946ee526581a52931dea/Makefile#L114-L115

andrejro, 2021-08-06:
One of the keys used to sign the Debian packages in the f-secure-foundry.github.io expired. You need to
add the new key:
```
wget https://f-secure-foundry.github.io/keys/gpg-andrej.asc -O - | apt-key add
```
This was also mentioned on the following mailing list post: https://groups.google.com/g/usbarmory/c/ExmqjGaGzFo

closed 2021-08-06 by andrejro