usbarmory/usbarmory-debian-base_image #50: Armory Mk 1 - uboot fails to start. Debian base image release 20210730

iam-TJ, 2021-08-14:
Earlier today I installed the `usbarmory-mark-one-usd-debian_buster-base_image-20210730` image to micro-SD card and found the Armory Mark 1 failed to boot. No sign of life, host PC doesn't see the CDC device, no heartbeat LED. Thinking I'd done something wrong I hacked about with checking SHA sums, including comparing the .raw image hash with the bytes on the micro-SD - all matched.
Eventually I thought to try an older image and found that `usbarmory-mark-one-usd-debian_buster-base_image-20210108` boots correctly.
This seems to suggest the boot-loader on the 20210730 image is broken.

iam-TJ, 2021-08-14:
I copied just the bootloader from the `20210108` image (the first 10240 sectors) to the micro-SD card with the `20210730` release and that boots, so the issue is in the bootloader.

One data-point - when it was failing the Armory was getting *very* hot very quickly, including the micro-SD card, which might suggest it was in a loop re-reading the card.

andrejro, 2021-08-17:
Recently U-Boot enforced the use of their "driver model" drivers. As a consequence the SD card configuration is taken from the devicetree file in `arch/arm/dts/imx53-usbarmory.dts`.
Such file is missing the card detect (CD) pin configuration, and therefore the card is not detected by the bootloader.

The following patch will be submitted to U-Boot to fix this issue:
```
diff --git a/arch/arm/dts/imx53-usbarmory.dts b/arch/arm/dts/imx53-usbarmory.dts
index f34993a490..433b62e736 100644
--- a/arch/arm/dts/imx53-usbarmory.dts
+++ b/arch/arm/dts/imx53-usbarmory.dts
@@ -91,6 +91,7 @@
 &esdhc1 {
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_esdhc1>;
+ broken-cd;
        status = "okay";
 };
```

digitalentropy, 2021-09-04:
Ran into the same issue today, and it took a while to debug because I thought I had a corrupt image somehow.

Any guidance on when the next pre-compiled release will be available? In the meantime, it seems it would be prudent to pull the last Mk 1 release since it is in effect defective.

andrejro, 2021-09-07:
The [20210907](https://github.com/f-secure-foundry/usbarmory-debian-base_image/releases/tag/20210907) release has just been published. This one correctly boots on Mk I devices.

closed 2021-09-07 by andrejro