usbarmory/usbarmory-debian-base_image #51: dns tool missing

Yalman, 2021-12-07:
Hello,

there are two little bugs.
First, the usbarmory-mark-two-*-base_image-20211129.raw.xz.sha256 files reference usbarmory-mark-two-*-base_image-20211130.raw.xz (show the date)
Second, in this image i miss a dns tool like "host" or "dig". At this moment i cant update or do anything depended from dns resolving.

best regards

abarisani, 2021-12-07:
Thanks for spotting the SHA256 error, we will fix it.

Concerning DNS tools you can install them with:

```
sudo apt install dnsutils
```

Yalman, 2021-12-07:
apt cant resolve domain names
```
root@usbarmory:/home/usbarmory# cat /etc/resolv.conf 
nameserver 8.8.8.8
```

```
root@usbarmory:/home/usbarmory# apt update
Err:1 http://security.debian.org/debian-security bullseye-security InRelease
  Temporary failure resolving 'security.debian.org'
...
```

abarisani, 2021-12-07:
This is unrelated to `dig` or `host` presence and more likely related to the fact that you are not routing the USB armory correctly from your host computer. Please see our [wiki](https://github.com/f-secure-foundry/usbarmory/wiki/Host-communication) for advice.

You have to enable your host computer connection sharing with the USB armory.

Yalman, 2021-12-07:
i did this and can ping google dns 8.8.8.8

abarisani, 2021-12-07:
Great, please close the issue if everything is solved from your side.

Yalman, 2021-12-07:
my new usbarmory was not in the correct firewalld zone. (ping worked, dns not)
sry for mistake and thank you for fast support.
best regards.

closed 2021-12-07 by Yalman