usbarmory/usbarmory-debian-base_image #52: SSH Host Key not properly generated

timkenhan, 2022-03-19:
I would like to bring to your attention that the host keys that are supposed to be generated during the sshd setup do not seem to be properly generated.

For now, creating it manually seem to fix the issue.

abarisani, 2022-03-21:
SSH keys are generated at first boot, we tested that there are no issues with this. It is not clear to me what you mean with "properly generated". Be aware that if you interrupt the first boot at generation (power loss, premature disconnection, etc.) that will leave you with corrupted keys as they won't be correctly re-generated at the following boot, maybe that's what happened in your case.

abarisani, 2022-03-28:
Please reopen if necessary.

closed 2022-03-28 by abarisani