usbarmory/usbarmory-debian-base_image #53: scc2 device-tree and modules don't seem to work

MonkeysAreEvil, 2022-07-15:
I have a Mk 1, running the 2022-07-06 Debian base image, and I'm trying to use the scc feature. Following the wiki, I boot with the scc2 device tree active (`ln -sf /boot/imx53-usbarmory-scc2.dtb /boot/imx53-usbarmory.dtb`), and then try and load the modules (`modprobe scc2; modprobe scc2_aes`). `scc2` seems to load correctly but `scc2_aes` fails with `modprobe: ERROR: could not insert 'scc2_aes': Operation not permitted`. Looking at the logs I see:

```
[ 4536.459961] scc2: loading out-of-tree module taints kernel.
[ 4536.497320] scc2: SCC declared UNIMPLEMENTED
[ 4536.506058] scc2: failed to set up interrupt handlers
[ 4536.529426] scc2: Driver Status is UNIMPLEMENTED
[ 4536.550834] scc2_aes: cannot read scm_version, aborting
```

which suggests that the device tree is not loading or working correctly. I've tried compiling the scc2.dts file, but this doesn't work either. Curiously, when I first tried to compile the dts file I got the following errors, which disappear on subsequent tries:

```
  DTC     arch/arm/boot/dts/imx53-usbarmory-scc2.dtb
arch/arm/boot/dts/imx53-usbarmory-scc2.dts:54.5-55.31: Warning (reg_format): /soc/aips@60000000/scc2@63fb4000:reg: property has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm/boot/dts/imx53-usbarmory-scc2.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
arch/arm/boot/dts/imx53-usbarmory-scc2.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm/boot/dts/imx53-usbarmory-scc2.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm/boot/dts/imx53-usbarmory-scc2.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm/boot/dts/imx53-usbarmory-scc2.dts:52.24-57.6: Warning (avoid_default_addr_size): /soc/aips@60000000/scc2@63fb4000: Relying on default #address-cells value
arch/arm/boot/dts/imx53-usbarmory-scc2.dts:52.24-57.6: Warning (avoid_default_addr_size): /soc/aips@60000000/scc2@63fb4000: Relying on default #size-cells value
```

Is there any way to confirm that the scc2 device tree is loading correctly, or could there be something else wrong?

abarisani, 2022-07-15:
Could you try to modify the dts as follows and test a dtb created with it ?

```
/ {
	soc {
		aips@60000000 {
			#address-cells = <1>;
			#size-cells = <1>;

			scc2: scc2@63fb4000 {
				compatible = "fsl,imx53-scc2";
				reg = <0x63fb4000 0x4000>,
				      <0x07000000 0x4000>;
				interrupts = <21 23>;
			};
		};
	};
};
```

MonkeysAreEvil, 2022-07-16:
Seems to compile fine, but booting with the corresponding dtb gives the same errors.

abarisani, 2022-07-18:
I just tested the Debian binary image for the Mk I and the scc driver works correctly with the following modification (already committed [here](https://github.com/usbarmory/usbarmory/commit/ffed1b813a9286e618fff585ee8682f0b7b9b542)):

```
/ {
	soc {
		bus@60000000 {
			#address-cells = <1>;
			#size-cells = <1>;

			scc2: scc2@63fb4000 {
				compatible = "fsl,imx53-scc2";
				reg = <0x63fb4000 0x4000>,
				      <0x07000000 0x4000>;
				interrupts = <21 23>;
			};
		};
	};
};
```

Please re-open if necessary.

Thanks!

closed 2022-07-18 by abarisani