usbarmory/usbarmory-debian-base_image #55: image-20230315 for MK1 is corrupted

VitalisDEV, 2023-03-29:
Hello

The image usbarmory-mark-one-usd-debian_bullseye-base_image-20230315.raw.xz  for MK1 is corrupted.
I tried 3 different cards, but the result was the same. MK1 does not start.
I take the previous image of usbarmory-mark-one-usd-debian_bullseye-base_image-20221114.raw.xz, then it works fine.

andrejro, 2023-03-30:
Thank you for the notification. The issue has been identified in the U-Boot image.
[This commit](https://github.com/u-boot/u-boot/commit/10f4f3f569976b7d8d0488b362ad5d97b3ca67f8#diff-e561840c2c9762ac3b5323d93855e8144f5e346b0618045987c5d510251cfa30L94) breaks the microSD card detection at boot.
We will fix in the next image release.

andrejro, 2023-07-21:
This has been fixed in the [20230721](https://github.com/usbarmory/usbarmory-debian-base_image/releases/tag/20230721) release.

closed 2023-07-21 by andrejro