usbarmory/usbarmory-debian-base_image #61: USB Armory Recognizing Devices

hiiii2341, 2024-10-08:
Hello! We have encountered a challenge while trying to get Video4Linux working with the USB Armory. Although the camera is detected, Video4Linux does not recognize any devices. We've tried using several different cameras but, unfortunately, haven't had any success so far.

If you have any advice or suggestions, we would be very grateful for your guidance. Thank you so much for your time and support, and I hope you have a great day.

Best regards,
Henry

abarisani, 2024-10-08:
Hello, it is likely that the required kernel modules are not compiled in.

Please ensure that the needed moduled are part of the kernel configuration, or compile your own kernel with the required module.

abarisani, 2024-10-08:
I am closing the issue as this is a V4L / Linux question and not something specific to the USB armory (it's a configuration issue, not a hardware one), if you still require support you might want to post a question (with relevant logs) to the [discussion group](https://groups.google.com/g/usbarmory).

closed 2024-10-08 by abarisani